brilliantlabsAR/frame-codebase #152: [feature request] Video input over Bluetooth

josuah, 2024-03-06:
This is a discussion to explore all the ways to implement video transfer over Bluetooth towards Frame display.

The 2 point that could throttle the transfer might be:
- Phone <---Bluetooth---> MCU communication
- MCU <---SPI---> FPGA communication

As well as amount of processing on the MCU and FPGA.

A few evoked solutions are:

- JPEG decoder counterpart to the encoder
- Use the bitmap library to stream the kind of data the FPGA can use, in 16-colors mode, updating the palette at every frame

josuah, 2024-03-06:
To illustrate what different resolution and color mode mean in practice:
None of these were confirmed to be possible yet!

400x400 full RGB888:
[image]

400x400 16 indexed colors:
[image]

200x200 full RGB888:
[image]

200x200 16 indexed colors:
[image]

100x100 full RGB888:
[image]

100x100 16 indexed colors:
[image]

josuah, 2024-04-02:
Here is what could be done for the Monocle:
[image]

Not yet tested on the hardware.

josuah, 2024-04-06:
This could be achieved on hardware on the Monocle
[image]

As seen above, still limitations that depend on the complexity of the image.

rohitrathnam, 2024-04-19:
Deep dive into how JPEG works: [Everything You Need to Know About JPEG Tutorial Series](https://youtube.com/playlist?list=PLpsTn9TA_Q8VMDyOPrDKmSJYt1DLgDZU4&si=FlCXh7XJDXtBtLwd)
Leaving this here for future reference

rmetchev, 2024-04-19:
Git repo to with the video series, used as reference for encoder.
https://github.com/dannye/jed

rohitrathnam, 2024-05-03:
Quick update, the EBR/LUT resources are almost full so it's unlikely we can support both JPEG decode and encode at the same time

josuah, 2024-05-04:
If two identical bitstreams can fit on the flash, this means it is possible to restart the FPGA at runtime with the alternative bitfile to cycle through features as the API uses them?

I think the LZO-compressed image was still quite large already though...

closed 2024-05-07 by siliconwitch

josuah, 2024-05-15:
Some feedback suggests to investigate the https://multimedia.cx/mirror/idroq.txt codec widely used on inexpensive (read: low-resource, low-complexity codec), high volume game consoles.

https://discord.com/channels/963222352534048818/984966420603482182/1240281670939050075