brilliantlabsAR/frame-codebase #169: [feature request] Extend character support (Chinese, Japanese, Korean, Brahmic, Arabic, Cyrilic, Hebrew...)

josuah, 2024-04-04:
This issue represents several requests from email support as well as Discord asking to increase the language support to more languages than those included by default.

Informations that would help:
- Planned or not
- Timeline, even without dates (which order)
- Workarounds (i.e. render an image) and libraries/snippets implementing them

rohitrathnam, 2024-04-19:
Connected to #114

josuah, 2024-04-20:
Connected to:
https://discord.com/channels/963222352534048818/1192416878413492285
https://discord.com/channels/963222352534048818/1172432945949196288
https://discord.com/channels/963222352534048818/1212777304137277470
and email discussions

josuah, 2024-04-24:
Requested today by email (Hebrew).

siliconwitch, 2024-05-07:
Including these as standard is not planned, but support will be coming soon for custom character sets to be loaded at runtime. Generate them using `frameutils` https://github.com/brilliantlabsAR/frame-utilities-for-python/blob/main/src/frameutils/create_sprites.py

More documentation around this once the feature is out

closed 2024-05-07 by siliconwitch

siliconwitch, 2024-05-07:
https://github.com/brilliantlabsAR/frame-codebase/issues/114

josuah, 2024-06-05:
Request for Cyrillic script:
https://discord.com/channels/963222352534048818/1247879872538349662

josuah, 2024-06-05:
To summarize: these languages are supported:
- Latin-based languages: alphabet + accented characters
 
These languages are not supported and support is not planned:
- Chinese, [1]
- Japanese, [1]
- Korean, [1]
- Brahmic families, [2]
- Arabic, [3] [2]
- Cyrilic, [4]
- Hebrew [3]

For the languages marked as [1] it will not be possible to support them as they are having too many characters for Frame
For the languages marked as [2], they will not be supported because they require combining characters not supported by the font engine
For the languages marked as [3], they will not be supported because they require right-to-left support, absent from Frame
For the languages marked as [4] it will be possible to upload a custom font with support for it when #114 gets completed, and this will allow them to work with Noa.
For the languages marked as [1], [2], [3], it will be possible to draw text on an image and send the image, but this will not work with Noa, only custom applications

theFestest, 2024-06-08:
For Japanese, in character count limited circumstances such as classic (style) video games or embedded systems, it is possible to write everything phonetically using hiragana (and optionally katakana). This does hurt legibility, but is common enough that people adapt and make more frequent use of spacing. Hiragana is ~50 symbols (plus variants/diacritics) and is sufficient to represent all sounds. Katakana is another ~50 (plus variants/diacritics) and would improve legibility relative to hiragana alone.

I would encourage considering if it is possible to support [these](https://loc.gov/marc/specifications/specchareacc/JapaneseHiraganaKatakana.html) Japanese phonetic characters, even if full kanji (Chinese character) support is not possible.

josuah, 2024-06-09:
> it is possible to write everything phonetically using hiragana (and optionally katakana)

Very good point! That would be possible to do with a custom font that also include hiragana and eventually katakana depending on our luck.

I updated the list above.

What remains is to make Noa answer only in hiragana, but this can be done by asking for it at the beginning.

[image]

theFestest, 2024-06-09:
That's great to hear. I will keep an eye on #114 then.

I don't have a Frame myself yet, but I would be interested to test out such a custom font when the time comes. Looking forward to it!

josuah, 2024-06-15:
Request by email for Japanese and Chinese.

josuah, 2024-06-20:
Request by email for Arabic and Hebrew.

josuah, 2024-06-23:
Request by email for Chinese.

josuah, 2024-06-27:
Requested by email for Japanese and Chinese.

josuah, 2024-07-08:
Request by email for Japanese

josuah, 2024-07-24:
Request by Discord for Arabic language:
https://discord.com/channels/963222352534048818/1242455966713253979/1265615049238646795