brilliantlabsAR/frame-codebase #182: [feature request] Turn dithering off on the ECX336CN

josuah, 2024-04-18:
Optionally turning dithering off would reduce the color space, in particular in low brightness situations, but is required when eye condition make dithering damageful.

To turn DITHERON off, the bit 1 of config 0x05 needs to be changed from true (as we configure it) to false (default):
https://github.com/brilliantlabsAR/frame-codebase/blob/main/source/application/display_configuration.h#L40

0xCB -> 0xC9

josuah, 2024-04-18:
If this is a customizable option, it need to either stay off by default, either have some way to store configuration in a persistent way that is not overwritten any time Noa is loaded.

JordansGithub, 2024-04-18:
This would be amazing for the large population of people who have screen sensitivity issues like me! If you can have a accessibility toggle to remove all forms of pixel flicker that would be huge for us screen sufferers🥹

josuah, 2024-04-18:
An alternative suggested by @JordansGithub by email is to add a command to change arbitrary display setting.

As long as none of them can cause hardware damage, this could allow all sorts of future tinkering without requiring to upgrade the firmware, in the same way as this was done: https://github.com/brilliantlabsAR/monocle-micropython/pull/281

rohitrathnam, 2024-04-19:
Tested this setting {0x05, 0xC3}, dithering on/off works: [PR](https://github.com/brilliantlabsAR/frame-codebase/pull/185)

josuah, 2024-04-20:
Wonderful reactivity! I mirror the gratitude from the user who requested it!

closed 2024-04-20 by josuah