brilliantlabsAR/frame-codebase #188: [feature request] Differentiate taps in the two directions from the IMU

josuah, 2024-05-02:
Frame currently detects taps in only one axis (say 'Y') regardless of the direction.

Allowing a +Y to a -Y to be differentiated could allow tapping on the left side and on the right side for different features, or eventually menu navigation.

Can this be implemented directly in Lua by constantly polling the IMU parameters?

Suggested from: https://discord.com/channels/963222352534048818/984966420603482182/1235682090721214534

rohitrathnam, 2024-05-03:
Polling the registers could be possible, but would likely take up a lot of resources. We don't yet have a queue system for i2c transactions so that would also block camera i2c which is needed for exposure and white balance control. Unfortunately the interrupt feature in the IMU only supports one direction per axis at a time 
[image]

josuah, 2024-05-04:
Thank you for the analysis! Hopefully the right poll frequency can be picked-up from Lua code... Which would require a timer() or sleep() or gettime() feature in Lua.

siliconwitch, 2024-05-07:
Yes, the datasheet is a little ambitious with this feature. It's quite awkward to dial in and not very reliable. Multidirectional taps should be possible with a simple filter and polling the IMU:
```lua
if frame.time.utc() - last_timestamp > 0.01 then -- every 10ms for example
    last_timestamp = frame.time.utc()
    data = frame.imu.raw()['accelerometer']
    -- do some filtering
    -- trigger action
end
```
This could also be started on a tap_callback to save some power and processor overhead

closed 2024-05-07 by siliconwitch