brilliantlabsAR/frame-codebase #219: lua api: display.clear could be added back

kjjohnsen, 2024-06-13:
There does not seem to be a way to clear the display without first putting some text into the buffer, and then displaying it.  So, it is possible to clear the display, but you have to issue a seemingly unnecessary text call to do it.  Here is an example:

```lua
frame.display.text('Hello world',50,100)
frame.display.show()
frame.display.text('',50,100) -- empty does not work
frame.display.show() -- does nothing
frame.display.text(' ',50,100) --non-empty works
frame.display.show() -- clears screen
```

Primary suggestion: re-add a clear function to the lua api
Secondary suggestion: make display.show with any empty buffer clear the display

Thanks!

siliconwitch, 2024-07-11:
Which firmware are you using? `frame.display.show()` should clear if no other draw command was given prior

kjjohnsen, 2024-07-11:
It was the latest at the time, the 
This part
`frame.display.text('',50,100) -- empty does not work
frame.display.show() -- does nothing`

Is what I would think is a bug/oddity, because it should clear.  I did not try to display.show() in a row.

CitizenOneX, 2024-08-02:
I get the same behaviour on firmware `v24.200.1206` - that is, the display isn't cleared if I only call `frame.display.show()` without any other draw calls.
Thanks!

CitizenOneX, 2024-10-08:
Update: Firmware v24.276.1359

Back-buffer clearing still seems to have a bug, and now the approach of "drawing a ' ' of text, then calling show()" doesn't work either.
When trying to clear the back buffer using a single space and show(), I got this (just the space character overlaid on the previous buffer). Calling show() twice in a row (with nothing in between) also doesn't clear the display, as before.
[image]

Other times the glitch shows up is just sending a break signal on wakeup - the "tap me in" prompt sometimes isn't cleared when the "Frame is Paired" text is shown. But other times it is - it's just intermittent.
[image]

So now it's not clear what would clear the display at all.
@siliconwitch said he'll try to take a look at this before long - thanks.