brilliantlabsAR/frame-codebase #229: PCB document available?

clawnotfound, 2024-07-07:
Hello, thank you for your outstanding contributions to the open-source community! I just wanted to know if the files for the PCB circuit board are open source? Where can I get them?

clawnotfound, 2024-07-07:
I understand that the PCB files involve some manufacturers' secrets, but could you release a PCB file that does not include the circuit for driving the AR display? I would like to develop a more affordable smart glasses based on this.

siliconwitch, 2024-07-11:
Hi there, as of now we don't have it available. This is exactly because of what you mentioned, but not just the display, but also camera and FPGA interface. What I can tell you is that the board is nothing too complicated. You'd be fine laying out from scratch using just the schematic