brilliantlabsAR/frame-codebase #262: IMU heading

tremblin, 2024-09-08:
IMU code for heading is missing (see TODO).

https://github.com/brilliantlabsAR/frame-codebase/blob/2e2a01ef91b7bc854987e0e7c35d117830ac3371/source/application/lua_libraries/imu.c#L170

Please complete the code for heading.

tremblin, 2024-09-20:
i've tried to calculate this on my own in LUA based on the frame.imu.raw() data. however the reading from the magnetometer seems to be wrong. it should read -|B| µT or +|B| µT if an axis is aligned to magnetic north (see https://eu.mouser.com/datasheet/2/821/MC6470_Datasheet_APS_048_0033v1_7_1-3003085.pdf, page 6), but i never get negative values from the readings!?

further your frame's coordinate system is a bit odd: i would envision the x-axis is pointing from the nose forward (direction of sight), y-axis to the right and z-axis down. in your case it seems the x-axis points left, y-axis backwards and z-axis downwards.

a good readup on axis + pcb placement and pitch/roll/yaw calculation can be found at https://www.mikrocontroller.net/attachment/292888/AN4248.pdf.

sshipman, 2024-11-02:
The AN4248 paper is an excellent resource for the algorithm, but it kind of makes hard iron estimation seem way more complex than it needs to be.  Basically, you can just move the device around in all sorts of orientations while recording min and max magnetic field ratings in all 3 dimensions, then the hard iron offsets are just the average in each dimension.
another resource:

https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

CitizenOneX, 2024-11-05:
Just in relation to the raw data readings from the magnetometer - they don't seem to be calibrated and adjusted with offsets in the sensor itself, but we can calibrate the values ourselves in the application and offset them - e.g. after calibration I used (-1112.50, -450.50, -6855.00) to be added to the raw values to get values ranging around 0 and sensible heading values.
[Example app](https://github.com/CitizenOneX/frame_imu)

In terms of implementing heading on Frame itself, it would seem that it would technically be possible to perform calibration (out of band with the request for heading, presumably!) and tilt-adjustment to calculate a heading value, but adjustments for declination would presumably need to take place in a connected app with a GPS and internet connection? Or, maybe all of it is done in-app if the preference is to keep Frame code lean?

zachary-kaelan, 2024-11-06:
Page 27 of the datasheet mentions "offset calibration" for the magnetometer, but has minimal information on how it works. Regardless, the offsets can be placed into registers manually and supposedly will be used when compensating the readings.