brilliantlabsAR/frame-codebase #284: Pink vertical lines on photos from Frame leads to bad image recognition performance

CitizenOneX, 2024-10-27:
At least as early as September 14th issues were reported:
[image]

As a result, numerous users are reporting that Noa is unable to make sensible descriptions of what it's seeing. I also see the issue in simple vision apps - this particular banding in the image really throws off the classifier models.

Back on August 6 it was definitely fine:
[image]

- The pink vertical lines issue looks to have been introduced with the **big camera firmare update on September 11:** (https://github.com/brilliantlabsAR/frame-codebase/releases/tag/v24.255.0646)

But with just this date range for completeness it could also possibly be:

- [Sep 4 Release v24.248.0928](https://github.com/brilliantlabsAR/frame-codebase/releases/tag/v24.248.0928)
- [Aug 28 Release v24.241.0836](https://github.com/brilliantlabsAR/frame-codebase/releases/tag/v24.241.0836)
- [Aug 15 Release v24.227.1307](https://github.com/brilliantlabsAR/frame-codebase/releases/tag/v24.227.1307)
- [Aug 8 Release v24.220.1445](https://github.com/brilliantlabsAR/frame-codebase/releases/tag/v24.220.1445)

siliconwitch, 2024-11-05:
Should now be fixed with the latest release. Thanks for the help with debugging!

closed 2024-11-05 by siliconwitch

PhilShaw852, 2024-11-14:
I have the same issue with the python SDK v1.2.2, will this be updated too?